MarcoMartines/GSM-GPRS-GPS-Shield #15: Library doesn't compile for ATMEGA 2560

mweber-ovt, 2015-02-20:
After changing the lines in the header file as prescribed for the MEGA:
//#define UNO
#define MEGA
the code doesn't compile. Error message:

Arduino: 1.0.6 (Mac OS X), Board: "Arduino Mega 2560 or Mega ADK"
In file included from /Users/michael/Dropbox/Arduino/libraries/SIM900/GSM.h:9,
                 from /Users/michael/Dropbox/Arduino/libraries/SIM900/SIM900.h:5,
                 from t1_Declarations.ino:10:
/Users/michael/Dropbox/Arduino/libraries/SIM900/WideTextFinder.h:11: error: ISO C++ forbids declaration of 'SoftwareSerial' with no type
/Users/michael/Dropbox/Arduino/libraries/SIM900/WideTextFinder.h:11: error: expected ';' before '*' token
/Users/michael/Dropbox/Arduino/libraries/SIM900/WideTextFinder.h:22: error: expected `)' before '&' token

Please help!
Thanks,
Michael

ingeltro, 2015-06-21:
Tengo el mismo problema ayuda please.

mweber-ovt, 2015-06-21:
Try Arduino 1.0.6 and not the newest version (1.6.x)

ingeltro, 2015-06-21:
Ok la verdad estoy muy complicado con esto pues no logró hacerlo funcionar
, muchas gracias por responder

ingeltro, 2015-06-24:
Funciono con arduino 1.0.6, muchas gracias.

mweber-ovt, 2015-06-24:
Happy to hear that!