MarcoMartines/GSM-GPRS-GPS-Shield #17: getIMEI return OK - not IMEI

antoniosap, 2015-03-06:
function:

int getIMEI(char* imei);

lost the imei received displyed in debug mode and return OK string

nishantpardamwar, 2016-03-31:
above fixed didn't work for me, so i wrote my own funtion.

replace these two function in SIM900.cpp

`int SIMCOM900::getIMEI(char *imei)`
`int SIMCOM900::SimpleRead()`

with

`int SIMCOM900::getIMEI(char *imei){
  //AT command to get IMEI.
  int i =0,a;
  unsigned long startTime = millis();
  char c;
  SimpleWriteln(F("AT+GSN"));
  while(true){
    a = SimpleRead();
    c = a;
    
    if(a!=-1){
      if(a == 13 && i==15){
        imei[15] = '\0';
        return 1;
      }

      if(a != 13 && a!= 10 && c!= 'O' && c!= 'K'){
        startTime = millis();
        imei[i] = c;
        i++;
      }
    }else if((millis() - startTime)>5000){
      return 0;
    }
  }
  return 0;
}`	

`int SIMCOM900::SimpleRead(){
  char datain;
  if(_cell.available()>0){
    datain=_cell.read();
    if(datain>0){
     return datain;
   }else
     return -1;
  }else
     return -1;
}`

michaelproino, 2016-06-16:
Hi,
Although I have faced with the same problem, I can not solve with this solution,
Can you help me please realted with getting IMEI with using Marco's library?