MarcoMartines/GSM-GPRS-GPS-Shield #20: Library doesn't work on MEGA 2560?

ronistiawan, 2015-04-20:
What I need to do to make this library work on Arduino MEGA 2560?

ronistiawan, 2015-04-22:
and now the problem is changed. It able to send and receive message, but always failed to connect to the internet. (The sim card is working perfectly on my phone).

bharathhs2290, 2015-05-20:
Dear ronistiawan are you now able to compile and work with this library on mega board?

ronistiawan, 2015-05-28:
Yes. I made some changes in the library. 
Thanks for your response.

bharathhs2290, 2015-05-28:
Can you please let me know what are those changes, it'll be of great help for me.

ronistiawan, 2015-06-02:
Open GSM.cpp and HWSerial.cpp. and do some modification showing below
GSM.cpp see line 37:
#ifdef MEGA
GSM::GSM(){
   _cell.begin(4800);
};

HWSerial.cpp see line 4:
#ifdef MEGA
   HWSerial::HWSerial(){
         Serial1.begin(4800);        
}

source: http://www.gsmlib.org/forum/viewtopic.php?t=38&f=8&sid=3c3ca15b4d0d24c034091c3c74f38cf5#p161

hope it helps you :)

closed 2015-06-13 by ronistiawan