MarcoMartines/GSM-GPRS-GPS-Shield #44: no matching function for call to 'SMSGSM::GetAuthorizedSMS(int, char [11], char [180], int, int, int)

boelle, 2016-09-20:
Hi

i use this lib and example 3 from http://www.open-electronics.org/how-to-send-and-receive-sms-with-gsmgprs-gps-shield/

sketch_sep18a:419: error: no matching function for call to 'SMSGSM::GetAuthorizedSMS(int, char [11], char [180], int, int, int)'

    stat=sms.GetAuthorizedSMS((int)pos,number,message,180,8,10);

my sketch: http://pastebin.com/ie806tp5

what have i done wrong here?

open-electronics do point to this lib so i assumed that their examples are correct.

boelle, 2016-09-25:
@madmaze sorry for pinging.... but do you have a clue? or know who might know what i do wrong

any pointers are welcome

inche-ali, 2016-10-02:
hello @boelle , I think your code should be like this..
`stat=sms.GetAuthorizedSMS((int)pos,number,strlen(number),message,180,8,10);`

boelle, 2016-10-08:
will try during the day, had a harddisk crash to take care of

boelle, 2016-10-08:
sket is now:

http://pastebin.com/9WW3kDFb

tried to solve some {} issues in the main loop but they are still wrong

boelle, 2016-10-08:
one thing that i never understand is this line:

char number[]="3921234567";

its in all the examples even thou example 3 i have based my code on is one that takes the number stored on the sim card pos 8-10... so why tell it a phone number?

inche-ali, 2016-10-08:
actually its not tell the phone number, it is a pointer to the reserved allocation for a string..

> phone_number: a pointer where the tel. number string of received SMS will be placed
>               so the space for the phone number string must be reserved

boelle, 2016-10-08:
oki doke

can you see how i could clean up the code more?

i still have trouble with {} in last void

inche-ali, 2016-10-09:
i believe your code missing one curly bracket  **}**  at the end of your code

boelle, 2016-10-09:
i posted the wrong link but its still the same: http://pastebin.com/raw/KYRJXCbQ

if i put a } at the very last line i get:

`
C:\Users\Bo\OneDrive\fyr\fyr.ino: In function 'void loop()':

fyr:393: error: expected '}' before 'else'

      else{

      ^

fyr:410: error: expected '}' before 'else'

      else{

      ^

C:\Users\Bo\OneDrive\fyr\fyr.ino: At global scope:

fyr:432: error: expected declaration before '}' token

 }}}

  ^

exit status 1
expected '}' before 'else'`

boelle, 2016-10-09:
if i put } before the last 2 else i'm left with:

fyr:432: error: expected declaration before '}' token

 }}

  ^

exit status 1
expected declaration before '}' token

but google is not my friend... or i'm stupid :-/

inche-ali, 2016-10-09:
wow what a code.. :) your  **if.. else** syntax is not right. 
you can use either 
```
 if (condition){
  //do something
}else if(condition){
   //do something
}else{
   //do something
}
```
or nested if.. else
```
 if (condition){
  //do something
}else {
  if(condition){
     //do something
   }else{
        if(condition){
            //do something
        }else{
           //do something
        }
   }
}
```
hope that will help

boelle, 2016-10-09:
will try...

i'm not any coding expert so i guess it will take a few tries to get right

boelle, 2016-10-09:
this is the code i came up with

http://pastebin.com/raw/Rhb4Tk2E

the logic as i see it or meant it to be:

first it will check for a new sms
then it will check if the sender is stored in pos 8-10 on the sim card
if the sender is in pos 8-10 it will check the word in the sms
if the word is start it should go to turnOn void
if the word is stop it should go to turnOff void
if the word is status1 then it should go to get_status void
if the word is status2 then it should go to get_status2 void
then it should delete the sms
and last if the word was start and it has been to the turnOn void it should now visit the keepAlive void 

this whole project is for my car.. it has a petrol driven preheater from webasto

the project is a merge of 2... one is the GSM/GPRS & GPS shield from open electronics, the other is a lib i found in this youtube video https://www.youtube.com/watch?v=6zZjSFef4yA

i dont use the LCD in that video but i use the lib from there and the description on how to connect the webasto heater...

instead of the lcd and buttons i use the GSM module so i can control it from anywhere and not limited by 100meter range on a good day

next step is to get the shield... or maybe i will hardwire direct to the mega.. i allready have the GSM module from a tidigino. if i go with direct wire it i just need the right level shifter for the serial line

and i also need to get the bits for connecting the webasto to the mega

so with some luck i will at least have the GSM connected before end of month

boelle, 2016-10-10:
Q: i could not find this in the examples

is it possible to monitor if the module is connected and automatic power it down and then power it up again?

i had some trouble last time i attempted this, each time i started the car the voltage up and down made it drop gsm connection.. and if i took power off the module and powered it up it worked fine until next time i started the car

skipperskip1970, 2019-01-19:
HI Bo.
Did you manage to get it running?
I would very much make the exact same project as you did. It is all I need, but I am not sure, how to do it all.

I have a GSM module (A6) and a Mega. I also have an OBD II  module hacked, so i can communicate with my heater.

I am missing instructions on how to do the wiring, and if there is any changes in some libaries.
I hope, that you will help me.

Svend 
Denmark.

boelle, 2019-01-19:
nope... and i have up on it many years ago

i got a new car also and i have decided not to put a heater in to it, its more trouble than its worth and the Peugeot 108 i got heats up so fast on its own that a heater will be a waste of money

closed 2019-01-19 by boelle